Sloeber/arduino-eclipse-plugin #1702: Arduino Mbed OS Platform and Boards import fails

dac1e, 2024-12-21:
sloeber	4.4.3.202402080117 on Windows 10.

[image]

When I try to import "Arduino Mbed OS" via Preferences/Platform and Boards, I get the error:
>>>>>>>>
Installing Arduino Board Platforms
Failed to install Arduino Mbed OS Nano Boards
Failed to extract tar.bz2. "C:/Sloeber-4.4.1/arduinoPlugin/downloads/ArduinoCore-mbed-nano-4.2.1.tar.bz2".
Can't extract file "C:\Sloeber-4.4.1\arduinoPlugin\packages\arduino\hardware\mbed_nano\4.2.1\.mbedignore", file already exists
<<<<<<<<
I tried that also for ArduinoCore-mbed-giga with the same result. 

However I can manually extract the bz2 file via 'mingw64 console: tar -xvf' and place the result in the  folder <Sloeber installation directory>\arduinoPlugin\packages\arduino\hardware\mbed_nano\4.2.1\. Everything works properly then.

jantje, 2024-12-21:
It states: "the file already exists".
Have you tried deleting the folder C:\Sloeber-4.4.1\arduinoPlugin\packages\arduino\hardware\mbed_nano\4.2.1 before installation?

dac1e, 2024-12-21:
Yes, I tried multiple times to delete the contents of the folder, but it didn't help. Even deleted the bz2 file, to force a fresh download. Didn't help either.--Gesendet mit der 1&1 Mail App

jantje, 2024-12-21:
Strange
I haven't seen anything similar.
I would try a new install (different location) and check my virus scanner

dac1e, 2024-12-21:
Thanks, I will do so tomorrow. However, it happens just on the mbed related platforms.--Gesendet mit der 1&1 Mail App