Sloeber/arduino-eclipse-plugin #1704: Icons are missing from the latest Linux download

emintz, 2025-01-11:
**Describe your environment**   
Do you use the Sloeber product or plugin?   I use the product.
Sloeber version number:   sloeber	4.4.3.202402070201	io.sloeber.product  
OS: Ubuntu 24.04
The url of the json that you used to download the platform: I pressed the download button on: https://eclipse.baeyens.it/stable.php?OS=Linux
The button url: https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_4_3_0/sloeber-ide-V4.4.3-linux.gtk.x86_64.tar.gz
  
A image of the project: should be attached

**Describe the bug**   
Icons are missing from the latest Linux install .tar.gz download

**To Reproduce**   
Steps to reproduce the behavior:   
1. Go to the Sloeber Latest Linux Stable page: https://eclipse.baeyens.it/stable.php?OS=Linux
2. Click on the green button labeled "Download sloeber-ide-V 4.4.3-linux.gtk.x86_64.tar.gz"
3. Run 'tar -x' on the file (tar -xf sloeber-ide-V4.4.3-linux.gtk.x86_64.tar.gz)
4. cd Sloeber
5. ls
6. Look for an 'icons' directory, which should contain the UI's icons. The directory does not exist.

I download the icons from https://github.com/Sloeber/arduino-eclipse-plugin/tree/master/io.sloeber.product.sdk/icons

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  
[image]

**Expected behavior**   

I expected to find Sloeber/icons containing the Sloeber icons to use on the desktop.

**Data to back up your claim**   

Run 'tar -tf' on the tar file mentioned above.

jantje, 2025-01-12:
I just tried it here (same version same ubuntu version) ,and all seems fine to me.
There is no icon folder but 
a)nor is there one on windows
b)the icons are there ands so is the splash screen.

emintz, 2025-01-14:
Sorry, I was not clear. The application's icons are 100% OK, no problems there. My issue is that Slober/icons is no longer included in the plugin. It contained:

```
16px-checkbox-checked.png    check.png             help.png          logo_32.png      save.png
16px-checkbox-unchecked.png  clear_console.png     learn.png         logo_48.png      serial.png
16px-upload.png              convertToSloeber.png  lens.png          new.png          ser_mon.png
arduino.png                  eclipse_lg.png        library.png       plotter.png      tape.png
arrowdown.png                filter_console.png    lock_console.png  preferences.png  timestamp_console.png
arrow.png                    folder.png            logo_16.png       sample.gif       toolbar_plotter.png
arrowup.png                  hammer.png            logo_256.png      save_dis.png     upload.png
```

I need `logo_256.png` for a desktop shortcut. Any chance we could add `logo*` to the package?

Thanks.

emintz, 2025-01-14:
Thanks. 

I'm having a hard time finding the UI .jar. The only obvious candidate I found,  ` ./plugins/io.sloeber.ui.nl1_4.4.3.202402070201.jar`, doesn't contain it. Can you advise, please?

jantje, 2025-01-14:
io.sloeber.ui_4.4.3.202402070201.jar contains a icon folder