Sloeber/arduino-eclipse-plugin #1705: String-valued C/C++ Argument Values Not Quoted

emintz, 2025-01-14:
Do you use the Sloeber product or plugin?   product
Sloeber version number: 4.4.3  
OS: Ubuntu LTS 24,04 x86
The url of the json that you used to download the platform: Download button on  https://eclipse.baeyens.it/stable.php?OS=Linux
A image of the project properties arduino showing the board settings: attached   
provide the name of the board and the platform version you are using.  Node32 ESP32S development board

[image]
of project properties->arduino  

`arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp` doesn't compile because the string-valued parameters ARDUINO_HOST_OS, ARDUINO_VARIANT, and ARDUINO_FQBN are not quoted. Quoting definitions in `arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/platform.sloeber.txt` fixes the problem. Fix attached.

To reproduce:

1. Install Linux Sloeber on Ubuntu 24.04 LTS.
2. Install version 3.0.7 ESP32 Board support.
3. Create a simple sketch.
4. Compile the sketch.
5. Compile fails

**Expected behavior**   
Sketch compiles correctly 

**Data to back up your claim**   
If applicable, add screenshots, console output and/or logs to help explain your problem.   

```
Building file: /home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp
Starting C++ compile
"/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++" -MMD -c "@/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/cpp_flags" -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_Node32s -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"Node32s\"" -DARDUINO_VARIANT="node32s" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="" -DARDUINO_FQBN="" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0    -DARDUINO_USB_CDC_ON_BOOT=0  "@/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/defines" "-I/home/eric/Documents/sloeber-workspace/blinky" -iprefix "/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/include/" "@/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/includes" "-I/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/dio_qspi/include" -I"/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32" -I"/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/node32s" -MMD -MP -MF"core/core/chip-debug-report.cpp.d" -MT"core/core/chip-debug-report.cpp.o" -D__IN_ECLIPSE__=1 "@/home/eric/Documents/sloeber-workspace/blinky/Release/file_opts" "/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp" -o "core/core/chip-debug-report.cpp.o"
/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp: In function 'void printSoftwareInfo()':
/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:211:67: error: expected primary-expression before ')' token
  211 |   chip_report_printf("  Compile Host OS   : %s\n", ARDUINO_HOST_OS);
      |                                                                   ^
/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp: In function 'void printBoardInfo()':
<command-line>: error: 'node32s' was not declared in this scope
/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:221:52: note: in expansion of macro 'ARDUINO_VARIANT'
  221 |   chip_report_printf("  Arduino Variant   : %s\n", ARDUINO_VARIANT);
      |                                                    ^~~~~~~~~~~~~~~
/home/eric/Sloeber.4.4.3/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:223:64: error: expected primary-expression before ')' token
  223 |   chip_report_printf("  Arduino FQBN      : %s\n", ARDUINO_FQBN);
      |                                                                ^
make[1]: *** [core/core/subdir.mk:312: core/core/chip-debug-report.cpp.o] Error 1
make: *** [makefile:104: all] Error 2
"make all" terminated with exit code 2. Build might be incomplete.```

**Additional context**   
Add any other context about the problem here.  

Fix attached
[platform.sloeber.txt](https://github.com/user-attachments/files/18411798/platform.sloeber.txt)

emintz, 2025-01-14:
Sorry, the fix didn't attach properly. Here it is.

[fix.txt](https://github.com/user-attachments/files/18411926/fix.txt)

jantje, 2025-01-14:
This is assumed to be fixed in V5. 
Try the latest nightly build here. https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/12759129694
Note that V4 and V5 are incompatible so you need to use the convert existing projects to sloeber project functionality (files->new->other->sloeber) to make things work.

emintz, 2025-01-14:
Thanks. Good to know. I'll upgrade to version 5 when it's GA.

I love Sloeber. Thanks for supporting it.

closed 2025-01-14 by emintz