Sloeber/arduino-eclipse-plugin #1710: V4.4.3 Core installer error

uzi18, 2025-02-11:
@jantje first try for a long time :

Win 10 Pro
Sloeber	4.4.3.202402080117	io.sloeber.product

Unzipped sloeber to C:\ 
Removed "Read only" attribute for Sloeber and all subfolders.
Launched 
Just want to install Uno R4 core:

[image]

Got error:

[image]

Closed sloeber, relaunched and selected Uno R4 like on first image:

[image]

```
Installing Arduino Board Platforms
Failed to install Arduino UNO R4 Boards
Failed to extract tar.bz2. "C:/Sloeber/arduinoPlugin/downloads/ArduinoCore-renesas_uno-1.3.2.tar.bz2".
Can't extract file "C:\Sloeber\arduinoPlugin\packages\arduino\hardware\renesas_uno\1.3.2\LICENSE", file already exists!
```

What do you think about it?
Anything to test for you? 

(edit)

Next found V5 is here so tried (074a1fd4b14b32e05e4cf811aa0b7a5d194353aa) and it works ok.

jantje, 2025-02-11:
I'm not sure why the install failed.
The second error is there because the install partially succeeded and Sloeber does not cleanup on failed installs.
Deleting the folder C:\Sloeber\arduinoPlugin\packages\arduino\hardware\renesas_uno\1.3.2 should do the trick.
However:
Sloeber downloads the archives in the downloads folder, unzips from there and keeps archive there as a download cache.
So it is a good idea to delete the downloads folder C:/Sloeber/arduinoPlugin/downloads just in case a failed download caused the original problem.

uzi18, 2025-02-11:
@jantje this is kind of note, will stick with V5. Sure noone will spend more time for V4.4. Thanks for answare.

closed 2025-02-11 by uzi18