Sloeber/arduino-eclipse-plugin #1712: V5 libraries path priority

uzi18, 2025-02-15:
Testing new Sloeber V5 and found a problem

For example on esp32 core 3.1.1 if project uses SD library when you click Apply&Close on project core properties.
Sloeber reassigns libraries, shows "Operation in progress" window.

Sets wrong lib - `ECLIPSE_HOME\arduinoPlugin\libraries\SD\1.3.0`
Here on esp32 core SD lib is provided as: `ECLIPSE_HOME\arduinoPlugin\packages\esp32\hardware\esp32\3.1.1\libraries\SD\src`
Shouldn't be core libs used in first place?
Or maybe already manually setted libs persistent?
Or both?

@jantje any advice? How to work with such problem?

jantje, 2025-02-15:
This sounds like a serious bug.
I'll try to reproduce this behavior (I'm not sure when I will have time though)

For your info: the intended behavior is
If a lib has been added to a project and the lib exists it remains persistent
If a lib has been added to a project and the lib does not exists it is replaced by the default lib.
The default lib is the library -stored under the folder which name is equal to the include directive -.h- in the private library location (in order as they appear in the Sloeber preferences); if matching private lib does not exists use a board library; else use latest installed managed library.

jantje, 2025-02-16:
I tried to reproduce and though I saw a issue I could not reproduce the behavior you described.
I created a ESP32 project and added the SD library from the board.
I went to project properties->Sloeber->selected apply close
The library is still the board library

[image]
however when I goto menu->Sloeber->Add library to project I see the SD listed both at the board as the managed libraries which is clearly wrong

[image]

uzi18, 2025-02-16:
@jantje  Thanks but maybe you can try to create project and not add the SD library from the board? 
Just open sloeber project properties apply & close.
This should be the case.

jantje, 2025-02-16:
If I do so no SD library is added.

uzi18, 2025-02-16:
Try to use example or code that use SD lib for esp32

uzi18, 2025-02-16:
> For your info: the intended behavior is If a lib has been added to a project and the lib exists it remains persistent If a lib has been added to a project and the lib does not exists it is replaced [...]

Interesting, never added manually lib to project. 
Is it also per project configuration? Will try this.

jantje, 2025-02-16:
>Is it also per project configuration? Will try this.

This is another big change between V4 and V5.
In V4 when changing the configuration all the libraries had to be "reattached" which means deleted and searched and attached.
In V5 each configuration has it's own folder for the core, variant and libraries. Therefore changing the configuration is just changing what is excluded from the build.
So in V4 you would add libs to the project.
in V5 it is to the configuration.

uzi18, 2025-02-16:
> if matching private lib does not exists use a board library; else use latest installed managed library.

Ok so for me this behavior not fully works as expected because esp32 lib is not used
Will do some test tomorrow

uzi18, 2025-02-18:
@jantje were right! 
Attach lib wizard works as expected :)
In fact both SD libs were marked there

jantje, 2025-02-18:
>In fact both SD libs were marked there

In the latest build they should not

closed 2025-03-31 by jantje