Sloeber/arduino-eclipse-plugin #1713: V5 #includes not resolved

PerennialNovice, 2025-02-17:
Not sure, whether this might be a known limitation....

When building the [Arduino-influx](https://github.com/tobiasschuerg/InfluxDB-Client-for-Arduino) example, the #include <WiFiMulti.h> does not get resolved correctly

WifiMulti.h is part of the WiFi library, but despite having the "automatically included libraries based on includes" an "automatically install missing libraries based on include?" activated, the WiFi lib is not added

sloeber 5.0.0.20251172146 on win11

jantje, 2025-02-17:
The list of known limitations is available here on github with following URL
https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue%20%20label%3A%22status%3A%20known%20limitation%22

This issue is indeed a known limitation (Sloeber assumes libname and foldername to be equal)
https://github.com/Sloeber/arduino-eclipse-plugin/issues/461

PerennialNovice, 2025-02-17:
thanks for clarifying, and the link!

jantje, 2025-02-17:
Please note that the link is available from the sloeber menu.