Sloeber/arduino-eclipse-plugin #1722: V5.0 some views are "inactive" after start

PerennialNovice, 2025-04-24:
**Describe your environment**   
WIN 11 24H2
latest sloeber product 5.0.0.202503302112

**Describe the bug**   
Right after starting sloeber, some views are "inactive", i.e. they are greyed out and don't work.
Right now, it is the serial monitor, progress and call hierarchy view

[image]

They are not updated and nothing happens with them.
Closing and reopening restores functionality

I think, but am not sure, that it is always one or all that are in focus after the sloeber start.

jantje, 2025-04-24:
Is this a recurring thing or only when opening Sloeber V4 workspaces?
I mean: in the process of upgrading V4 to V5 it is possible that some views have wrong ID's and are frozen/greyed.
However once the ID's have been upgraded in the workspace they should not revert to greyed.

PerennialNovice, 2025-04-25:
It is happening on every start of V5.
Up to now, in V5 I have only projects created in V5.

jantje, 2025-09-27:
Is this still a issue?

PerennialNovice, 2025-09-29:
Just started version 5.0.0.202506091417

Yes, still some views (call hierarchy, serial monitor, progress) "greyed out" right after startup.

[image]

changed serial monitor to console, closed sloeber and re-opened.
Now console is greyed out.
It somehow seems to be the last ones open in the lower views

jantje, 2025-10-07:
I have the same behaviour now.
The view with the focus (= last one open) seems to fail.
I also get a stacktrace
```
org.eclipse.ui.PartInitException: Abnormal Workbench Condition
	at org.eclipse.ui.internal.WorkbenchPage.showView(WorkbenchPage.java:4253)
	at org.eclipse.ui.internal.WorkbenchPage.showView(WorkbenchPage.java:4225)
	at io.sloeber.ui.actions.OpenSerialMonitorHandler.execute(OpenSerialMonitorHandler.java:36)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:277)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:98)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:103)
	at java.base/java.lang.reflect.Method.invoke(Method.java:580)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:305)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:237)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:174)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:165)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:485)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:204)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:444)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:472)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:494)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:91)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4362)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1214)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4160)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3748)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1151)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1042)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:678)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:583)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:185)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:219)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:149)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:115)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:467)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:298)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:103)
	at java.base/java.lang.reflect.Method.invoke(Method.java:580)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:627)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:575)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1431)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1403)
```
As far as I can see the field object of the PartImpl related to the "bad view" is null.
Why that is ... I have no clue.

PerennialNovice, 2025-10-07:
It also happens on a completely fresh install of the latest build with no project in a new workspace

jantje, 2025-10-07:
Strange; since I deleted these folders I can no longer reproduce the issue and it worked every time before.

jantje, 2025-10-07:
I was smart enough to save the .metadata and when I put that one back the problem reappears.
Do you use oomph as that may undo part of the delete

[image]

PerennialNovice, 2025-10-08:
nope, nothing found with "oo" in preferences

[image]