Sloeber/arduino-eclipse-plugin #1726: Build fails after Git share

pcarew, 2025-08-10:
I found a previous bug report for this (https://github.com/Sloeber/arduino-eclipse-plugin/issues/748),  but there didn't appear to be a resolution for it.

I can reproduce this with just a **standard** Sloeber project
Environment:
Windows 10
Eclipse: Version: 2020-06 (4.16.0)
Sloeber: Eclipse Arduino IDE (plugin version)	4.3.3.202005042010
Arduino 1.8.3, Uno

**To Reproduce:**
Click New Project ->sloeber sketch -> default ino

**compile (verify**) -> All ok
**project -> clean**
**create .gitignore** with Release/\*, core/\*, Archive/\*, core/\* .settings, .cproject, .project, spec.d, 

**team -> share**
Project is copied correctly to git location

**Compile (Verify)**: failed at Archiver point:

Starting archiver'
"C:\eclipse\jee202006_416\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs     C:\Users\PaulCarew\workspace-cdt\ArduinoFlowMon\Release/arduino.ar  .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o 
c:\eclipse\jee202006_416\arduinoplugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: **C:\Users\PaulCarew\workspace-cdt\ArduinoFlowMon\Release/arduino.ar: No such file or directory**
makefile:73: recipe for target 'C:\Users\PaulCarew\workspace-cdt\ArduinoFlowMon\Release\arduino.ar' failed
make: *** [C:\Users\PaulCarew\workspace-cdt\ArduinoFlowMon\Release\arduino.ar] Error 1
"C:/eclipse/jee202006_416/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

It looks like it is still trying to use the workspace location for arduino.ar

I'm sure I must be doing something wrong, but this is a perplexing issue.

Thanks Jante.
Paul.

jantje, 2025-08-10:
After sharing the project you need to open the project properties->sloeber and select apply/ok because stuff has moved and sloeber doesn't know.

pcarew, 2025-08-10:
Jantje,
I've been using and appreciating your Arduino plugin for a while now. I'd like to donate a contribution to you to say thank you.
Are you accepting and donations at the present time?