Sloeber/arduino-eclipse-plugin #1728: How to enable sprintf support for floats in Teensy 4.1 builds

cabintech, 2025-09-24:
Googled all over and could not find how to do this in Sloeber. By default, sprintf() using %f formatting fails (prints nothing). Seems I need to add

`-u _printf_float`

to a linker option somewhere but I cannot see where to add this in the project properties. The instructions I found don't seem to apply:

_To enable floating-point formatting with sprintf for a Teensy 4.1 in the Sloeber IDE, you must add a specific linker flag to your project settings. The ARM toolchain's newlib library, which is used by default, has floating-point printf support disabled to save space. 
Add the linker flag in Sloeber:
1 Right-click on your project in the Sloeber Project Explorer and select Properties.
2 In the Properties dialog, navigate to C/C++ Build -> Settings.
3 Under the Tool Settings tab, expand the MCU ARM C++ Linker section (or MCU ARM C Linker if it's a C project).
4 Click on Miscellaneous.
5 In the Linker flags text box, add the following flag: -u _printf_float.
6 Click Apply and then Close._ 

I don't see any "MCU ARM C++ Linker" in the Tool Settings tab...

jantje, 2025-09-24:
Can you provide a link to the 
`The ARM toolchain's newlib library, which is used by default, has floating-point printf support disabled to save space.`

As far as I know Teensy 4.1 is a 64 bit processor which means 64 bit float is supported on top of the "normal" 32 bit float and that may create some confusion.

jantje, 2025-09-24:
>Sorry for all the bother, never would have guessed that a compile optimization would null out this kind of basic functionality

float and String ... I can see how they get dumped for smallest code.

The good news is
1) we have become a little bit smarter
2) Your problem is fixed.

closed 2025-09-24 by jantje